linuxwacom/input-wacom #394: insmod - could not insert module, invalid module format

Lexpeartha, 2023-09-29:
Executing `sudo insmod 4.5/wacom.ko` started giving:
```
insmod: ERROR: could not insert module 4.5/wacom.ko: Invalid module format
```
which started when I upgraded to `Linux 6.5.5-200.fsync.fc38.x86_64` kernel version

Pinglinux, 2023-09-30:
Also, why did you need to load the wacom driver yourself? What was wrong with the system in-box driver?

Lexpeartha, 2023-09-30:
> Did you rebuild wacom.ko after the upgrade? Your wacom.ko was built for the older kernel.

That fixed it, thank you

> Also, why did you need to load the wacom driver yourself? What was wrong with the system in-box driver?

Not sure, but my device is not recognized unless I execute this command myself on every boot, which is what I've been doing

closed 2023-09-30 by Lexpeartha