linuxwacom/input-wacom #396: Touch screen stopped working (stylus works) on Thinkpad X1 Yoga 4gen

shaked8634, 2023-10-18:
Using Gnome + Wayland
Wacom Driver version 2.0.0
I tried with 2 kernels: 6.5.7 and LTS 6.1.58. 

I can see:  **wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.** like in #342.
```shell
# libwacom-list-local-devices                                                                                                                                                  [0]
devices:
- name: 'ISDv4 51B7'
  bus: 'usb'
  vid: '0x056a'
  pid: '0x51b7'
  nodes: 
  - /dev/input/event6: 'Wacom Pen and multitouch sensor Pen'
  - /dev/input/event5: 'Wacom Pen and multitouch sensor Finger'
```
```shell
# journalctl -k --grep wacom --no-pager
Oct 18 11:30:33 localhost kernel: usb 1-6: Manufacturer: Wacom Co.,Ltd.
Oct 18 11:30:33 localhost kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor Touchscreen as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:51B7.0001/input/input6
Oct 18 11:30:33 localhost kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:51B7.0001/input/input7
Oct 18 11:30:33 localhost kernel: hid-generic 0003:056A:51B7.0001: input,hiddev96,hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input0
Oct 18 11:30:33 localhost kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor Stylus as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:51B7.0002/input/input8
Oct 18 11:30:33 localhost kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:51B7.0002/input/input9
Oct 18 11:30:33 localhost kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:51B7.0002/input/input10
Oct 18 11:30:33 localhost kernel: hid-generic 0003:056A:51B7.0002: input,hiddev97,hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input1
Oct 18 11:30:35 localhost kernel: input: Wacom Pen and multitouch sensor Finger as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:51B7.0001/input/input20
Oct 18 11:30:35 localhost kernel: wacom 0003:056A:51B7.0001: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input0
Oct 18 11:30:36 localhost kernel: input: Wacom Pen and multitouch sensor Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:51B7.0002/input/input22
Oct 18 11:30:36 localhost kernel: wacom 0003:056A:51B7.0002: hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input1
Oct 18 11:36:34 localhost kernel: wacom 0003:056A:51B7.0002: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Oct 18 11:36:36 localhost kernel: wacom 0003:056A:51B7.0002: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Oct 18 11:36:38 localhost kernel: wacom 0003:056A:51B7.0002: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Oct 18 11:36:40 localhost kernel: wacom 0003:056A:51B7.0002: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Oct 18 11:36:44 localhost kernel: wacom 0003:056A:51B7.0002: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Oct 18 11:36:49 localhost kernel: wacom 0003:056A:51B7.0002: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Oct 18 11:36:50 localhost kernel: wacom 0003:056A:51B7.0002: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Oct 18 11:36:50 localhost kernel: wacom 0003:056A:51B7.0002: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Oct 18 11:36:53 localhost kernel: wacom 0003:056A:51B7.0002: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out
```

I also posted on [Arch Linux forum](https://bbs.archlinux.org/viewtopic.php?pid=2126845#p2126845).
Attached hid-recorder for all devices while I'm touching the screen.
[hid-recorder_hidraw0.txt](https://github.com/linuxwacom/input-wacom/files/13030222/hid-recorder_hidraw0.txt)
[hid-recorder_hidraw1.txt](https://github.com/linuxwacom/input-wacom/files/13030223/hid-recorder_hidraw1.txt)
[hid-recorder_hidraw2.txt](https://github.com/linuxwacom/input-wacom/files/13030224/hid-recorder_hidraw2.txt)

Please let me know if anything is missing here and appreciate any help and thanks in advance :)

shaked8634, 2023-10-19:
It's started to work again without any known change I made :person_bald:  :gun:

shaked8634, 2023-10-21:
Reopend since it happen sporadically.

I can see this in dmesg:

> hid-sensor-hub 001F:8087:0AC2.0005: timeout waiting for response from ISHTP device

lejono, 2023-11-07:
I have this problem as well. And the cursor jumps around, causing constant havoc.  There are many reports but I haven't seen a solution eg: https://github.com/linuxwacom/input-wacom/issues/342 https://github.com/linuxwacom/input-wacom/issues/262 https://github.com/linuxwacom/input-wacom/issues/288

Originally, I thought it was because I was trying to use the lenovo or wacom active pen, but this also happens with the lenovo stylus which comes with the laptop (x1 yoga 5th gen).

modinfo wacom | grep version
version:        v2.00
srcversion:     CE2807BA4D6A356047D60C4
vermagic:       6.2.0-35-generic SMP preempt mod_unload modversions

lejono, 2023-11-07:
Here is the output of libinput record for the stylus.
[libinput_record_stylus.txt](https://github.com/linuxwacom/input-wacom/files/13283637/libinput_record_stylus.txt)

jigpu, 2023-11-09:
This issue has been around for a while, but I don't believe that its root cause is a driver issue (meaning that it can be very difficult to "fix" from the driver). There are too many instances of the bug appearing and disappearing without any obvious change to the affected system; let alone changes that I would expect to influence this.

I'll follow-up on some of the wilder possibilities of what could be going wrong, however.

I've made four branches that disable various parts of the driver and am interested in knowing if any of them have any effect on the jumping issue. You'll first need to clone my repository as follows:
~~~
$ git clone https://github.com/jigpu/input-wacom.git
$ cd input-wacom
~~~
Next you'll need to check out one of the four test branches:
* `$ git checkout stylus-jump-A`
* `$ git checkout stylus-jump-B`
* `$ git checkout stylus-jump-C`
* `$ git checkout stylus-jump-D`

Compile and install the code as per the wiki instructions and then reboot. Run `cat /sys/module/wacom/version` to verify what version of the driver is actually running. It should end with either `g7e1c20e` for A, `g025759f` for B, `geaeb441` for C, or `gd449720` for D. You can now use the modified driver for some time and see if it has any effect on the jumping behavior. If there doesn't seem to be any effect, try checking out the next branch and then repeat these steps again with it.

jigpu, 2023-11-30:
Checking to see if there's any update on this.

jinek, 2024-06-09:
I had styles cursor jumping around issue too, it was strangely solved when I installed this LENOVO firmware update https://download.lenovo.com/pccbbs/mobiles/n3arl02w.zip manually.