linuxwacom/input-wacom #399: Bamboo One CTE-460 not being loaded with correct driver, possible driver failure?

vertesians, 2023-10-30:
I'm on Arch Linux using awesomewm on X, with `libinput 1.24.0-1`, `xf86-input-libinput 1.4.0-1`, `libwacom 2.8.0-1`, and `xf86-input-wacom 1.2.0-1` installed. Sometime in the past month or two, my drawing tablet started being loaded with the wrong driver:

```
[~] lsusb | grep -i "wacom"
Bus 001 Device 006: ID 056a:006a Wacom Co., Ltd CTE-460 [Bamboo One Pen (S)]
[~] sudo dmesg | grep -i "wacom"
[  122.492438] usb 1-3: Manufacturer: Wacom Co.,Ltd.
[  122.498700] input: Wacom Co.,Ltd. CTE-460 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:056A:006A.0004/input/input19
[  122.498970] hid-generic 0003:056A:006A.0004: input,hidraw3: USB HID v1.00 Mouse [Wacom Co.,Ltd. CTE-460] on usb-0000:00:14.0-3/input0
```

Looking at journalctl, a bunch of warnings show up when I plug the device in. Seems like the wacom driver is failing when it tries to load?

```
Oct 30 12:38:17 raclette kernel: usb 1-3: new full-speed USB device number 5 using xhci_hcd
Oct 30 12:38:17 raclette kernel: usb 1-3: New USB device found, idVendor=056a, idProduct=006a, bcdDevice= 1.00
Oct 30 12:38:17 raclette kernel: usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Oct 30 12:38:17 raclette kernel: usb 1-3: Product: CTE-460
Oct 30 12:38:17 raclette kernel: usb 1-3: Manufacturer: Wacom Co.,Ltd.
Oct 30 12:38:17 raclette kernel: input: Wacom Co.,Ltd. CTE-460 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:056A:006A.0004/input/input19
Oct 30 12:38:17 raclette kernel: hid-generic 0003:056A:006A.0004: input,hidraw3: USB HID v1.00 Mouse [Wacom Co.,Ltd. CTE-460] on usb-0000:00:14.0-3/input0
Oct 30 12:38:17 raclette mtp-probe[1158]: checking bus 1, device 5: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Oct 30 12:38:17 raclette mtp-probe[1158]: bus: 1, device: 5 was not an MTP device
Oct 30 12:38:17 raclette (udev-worker)[1157]: 1-3: Process 'check_driver wacom /devices/pci0000:00/0000:00:14.0/usb1/1-3 usb' failed with exit code 1.
Oct 30 12:38:17 raclette (udev-worker)[1157]: 1-3:1.0: Process 'check_driver wacom /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0 ' failed with exit code 1.
Oct 30 12:38:17 raclette (udev-worker)[1157]: 0003:056A:006A.0004: Process 'check_driver wacom /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:056A:006A.0004 ' failed with exit code 1.
Oct 30 12:38:17 raclette (udev-worker)[1157]: input19: Process 'check_driver wacom /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:056A:006A.0004/input/input19 usb' failed with exit code 1.
Oct 30 12:38:17 raclette (udev-worker)[1168]: mouse2: Process 'check_driver wacom /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:056A:006A.0004/input/input19/mouse2 usb' failed with exit code 1.
Oct 30 12:38:17 raclette (udev-worker)[1165]: hidraw3: Process 'check_driver wacom /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:056A:006A.0004/hidraw/hidraw3 ' failed with exit code 1.
Oct 30 12:38:17 raclette root[1173]: ACPI group/action undefined: button/up / UP
Oct 30 12:38:18 raclette (udev-worker)[1157]: event7: Process 'check_driver wacom /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:056A:006A.0004/input/input19/event7 usb' failed with exit code 1.
Oct 30 12:38:18 raclette mtp-probe[1175]: checking bus 1, device 5: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Oct 30 12:38:18 raclette mtp-probe[1175]: bus: 1, device: 5 was not an MTP device
```

(All lines with "udev-worker" are warnings.)

The tablet was working normally a few months ago, so I'm not sure what went wrong.

Pinglinux, 2023-11-30:
Not sure why hid-generic picked up your Bamboo. Please share what you see from dmesg after issuing:

$ sudo rmmod wacom
$ sudo modprobe wacom

Pinglinux, 2024-05-17:
It is not a driver issue.

closed 2024-05-17 by Pinglinux