linuxwacom/input-wacom #410: Consider Disabling AES Pens on Broken Platforms

Doomsdayrs, 2024-02-06:
It seems that the AES pens for tablet PC devices are currently extremely hard to use.

Thus, dropping support for the devices until the root problem is determined and solved might be the best course in order to provide a stable user experience.

## Steps

1. Disable support for the malfunctioning device(s) by default
2. Print error / warning message into logs in regards to the device(s) being unsupported
3. Provide kernel parameter to re-enable the malfunctioning device(s) for testing purposes.

## Related issues
#409 
#342

Pinglinux, 2024-03-10:
From the history, you can tell that we did our best to figure out a solution for the issue. All information we gathered proves that the root cause is not in the tablet or our driver. It is an integrated system issue.

I like your suggestion of disabling the events in the kernel driver. We'll do that in input-wacom, instead of upstream, since it will be a PID specific solution. I'd like to know how many people are willing to build and install the driver before schedule effort for the workaround.

mdarnold1, 2024-03-26:
As an affected user I would very much prefer that this NOT go ahead, as it makes the nature of the problem less obvious to the user, and harder to diagnose and reverse.  I specifically purchased my hardware for stylus functionality, so I was disappointed to find kernel regressions that seem to have made the stylus less performant, and I'm still looking for solutions. If disabling were to go ahead, making the results of steps 3 & 2 very visible to the user should be a priority.

Pinglinux, 2024-04-03:
@Doomsdayrs Looks like not everyone wants the same workaround. That doesn't stop you from disabling the pen for your device.  Follow one of the following links will disable the stylus:

[1] https://askubuntu.com/questions/927022/how-can-i-disable-touchscreen-while-using-wayland
[2] https://github.com/linuxwacom/input-wacom/wiki/Fixed-device-files-with-udev

closed 2024-05-17 by Pinglinux