linuxwacom/input-wacom #412: Wacom driver won't load properly

winteriscariot, 2024-02-16:
Arch Linux

I have an Intuous BT S that i'm trying to get to function to no avail. I've tried both via USB as well as bluetooth but it appears that it's getting picked up by hid-generic:

```
# dmesg | grep -i wacom
[    4.276210] usb 1-1.4.3: Manufacturer: Wacom Co.,Ltd.
[    4.574596] hid-generic 0003:056A:0376.0006: hiddev96,hidraw4: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos BT S] on usb-0000:02:00.0-1.4.3/input0
```

the device shows up in lsusb as such:

`Bus 001 Device 006: ID 056a:0376 Wacom Co., Ltd CTL-4100WL [Intuos BT (S)]`

however nothing functions on the tablet. while lsusb shows that it exists, the pen doesn't move the cursor at all. i have the xf86-input-wacom driver as well as the libwacom package installed:

```
# pacman -Q |grep wacom
libwacom 2.10.0-1
xf86-input-wacom 1.2.0-1
```

not entirely sure where to go from here. the Graphics Tablet section of the Arch Wiki did not provide me with anything useful

thanks!

Pinglinux, 2024-02-28:
Weird. What version of "Arch Linux" do you run? What's the kernel version? If it is newer than 5.x, it should work out of box. By any chance, did you try to install wacom.ko from input-wacom?

jigpu, 2024-02-29:
Please also perform the following steps for us:
1. Run `modinfo hid_generic` and copy/paste the output here
2. Run `modinfo wacom` and copy/paste the output here
3. Download our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh), run it, and attach the generated tarball here

winteriscariot, 2024-02-29:
I resolved the issue, it was not related to the driver itself.

For anyone coming across this, 2 things:

1) a bad USB cable: switching cables allowed the tablet to work properly
2) blueman wasn't properly connecting the tablet to the input service; changing to blueberry (ie, gnome-bluetooth) fixed this

sorry for taking people's time, the collision of those 2 issues at the same time made me feel like there was a deeper issue.

closed 2024-02-29 by winteriscariot