linuxwacom/input-wacom #418: Linux multi-monitor pen tracking problem

walthaas, 2024-04-21:
I run Fedora 39 with XFCE and a Wacom Cintiq Pro 16. When I add the Wacom as a monitor, adjust the resolution to match my monitor and check "Mirror displays" in the Display settings, everything works as expected. But this is a little inconvenient because I need to switch between workspaces to use any software besides the graphics software (Blender, Krita) I'm using for my application.

So, in an attempt to be able to use both screens simultaneously, I turned off the display mirroring and configured the Wacom to the right of the monitor. Well.... it almost works. I can have the graphics software display on the Wacom while the monitor is usable for the browser or whatever. When I move the mouse it moves nicely between the monitor and the Wacom and tracks just as expected. But when I pick up the Wacom pen, the dot tracks the pen at the right side of the tablet, but when I move the pen towards the left, the dot moves twice as fast as the pen so it isn't under the pen, so the dot reaches the left edge of the tablet when the pen is in the middle. This tracking problem doesn't depend on the graphics application, it occurs on an empty screen, and it doesn't happen with the mouse, only with the Wacom pen. My conclusion is that the problem is somewhere in Wacom specific code between detected pen position and display on the Wacom screen. My GPU is RTX3080, but I can't see why that would affect the pen but not the mouse.

I haven't found any other report of this problem. Does anyone know of a fix, or am I doomed to mirror displays forever?

I've never worked on the Wacom driver code, but thinking about the problem my guess would be that the driver uses the combined width of the two monitors when it computes where to place the dot, rather than the width of only the monitor showing the tablet. At least that would seem to explain the behavior.

Pinglinux, 2024-05-13:
It sounds like the system takes the tablet as an opaque tablet. I assume the pen could reach the monitor on the left, if you move the pen pass the middle of the tablet screen. Am I right? Did you try the switch monitor feature in "Wacom Tablets" under Gnome Settings?

walthaas, 2024-05-13:
Correct, if I move the pen to the middle of the tablet the cursor crosses to the monitor.  When the pen is at the left edge of the tablet the cursor reaches the left edge of the monitor.

I'm using XFCE so don't have access to Gnome settings.  In XFCE, the Mouse and Touchpad popup shows four Wacom devices.  Tinkering with the settings (Tracking mode, Rotation) I couldn't find anything helpful.  I tried all four Wacom devices and got the same results with all.

In the Display Settings popup I can drag the tablet and monitor to opposite positions.  Of course that puts the graphic application on the monitor not the tablet so I can't use it, but when I pass the pen across the tablet it behaves the same - at the midpoint the cursor switches to the monitor, at the left edge of the tablet the cursor reaches the left edge of the monitor.

Pinglinux, 2024-05-17:
I forgot you are using XFCE. The root cause is your configuration UI considered your Cintiq Pro as an opaque tablet. So, to the UI, it needs to map the pen/tablet to the whole desktop, which consists of two monitors. The issue is beyond this project's scope, which is for Wacom kernel driver. Please post your issue to the relevant pages at https://xfce.org.

Thank you for reaching out to us and for using Wacom tablet on Linux!

closed 2024-05-17 by Pinglinux