linuxwacom/input-wacom #421: Incorrect indentation in `configure`

NeumoNeumo, 2024-05-07:
In the v1.12.0 release, the `configure` file has incorrect indentations from line 4278 to line 4286:
`/usr/src/linux\-$X/lib/modules/$X/source/.config` should be in the same line but they appear to lie on two lines in the code. I tried but failed to find out what's going on beneath because of my unfamiliarity with `autoconf`. 
[image]

After deleting the extra line breaks, `configure` runs successfully.
[image]

Pinglinux, 2024-05-17:
I assume you are talking about v1.2.0 since there is no v1.12.0 release. I compared configure from v1.1.0 and v1.2.0. There is no difference between those two files. To understand your issue, I need more details about your system and the actions you took to see the issue.

Which Linux distribution do you use? What kernel version are testing with? What error message do you see when you run ./configure?

./configure works on my Fedora 40, the latest, without problem.

NeumoNeumo, 2024-05-17:
1. Sorry for my typo. It is v1.2.0 actually. (I have corrected it in the original text)
2. I have changed my kernel from 6.6 to 6.5, and now everything is right.

closed 2024-05-17 by NeumoNeumo