linuxwacom/input-wacom #424: Ubuntu 22.04.4 LTS x86_64, 80VV Lenovo MIIX 720-12IKB, My tablet cannot touch anymore

AnthraxReaper, 2024-06-03:
I have installed input-wacom version 1.2.0 because the stock kernal would make the touchscreen buggy (example : touch sometimes will not register, will have 'ghost' input). When I reboot the device, the touchscreen seems to not working anymore. When I check the version of the kernal driver:
`grep "" /sys/module/wacom*/version`
the output is :
`v2.00-input-wacom`
which is weird

Also, when I try to troubleshoot the problem:
`sudo rmmod wacom
sudo rmmod wacom_w8001
sudo modprobe wacom
sudo modprobe wacom_w8001`
The output is:
`rmmod: ERROR: Module wacom_w8001 is not currently loaded`

By the way, the name of the touchscreen is Wacom ISDv4 5099

Pinglinux, 2024-06-04:
I built and installed 1.2.0. And I got v2.00-input-wacom too. We need to look into that version number issue. But, your driver was built properly. Please try: 

sudo rmmod wacom
sudo modprobe wacom  (or sudo insmod 4.5/wacom.ko)

to see what happens. You do not need wacom_w8001.

Since you are using a Lenovo laptop, I feel it is most likely not a driver issue.

AnthraxReaper, 2024-06-22:
Sorry for late report. But I have found the solution for the problem, which is dirty screen. Unfortunately the screen is prone to ghost touch and/or unresponsiveness if the screen is dirty.

closed 2024-06-22 by AnthraxReaper