linuxwacom/input-wacom #428: Pen starts to have long delays after some usage, dmesg says wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.

1over137, 2024-06-09:
Device: Thinkpad X1 Yoga 6th gen
Stylus: Lazarite AES pen, builtin pen (same behavior)
OS/kernel: Gentoo, Linux 6.6.13

Normally the pen delay is relatively small (although still slower than on Windows), but after a while of usage it becomes much slower (cursor lagging behind). When this behavior occurs, each approach of the pen is accompanied by this message in dmesg:

`wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.`

Reboot fixes it temporarily

LDprg, 2025-03-18:
I have the same problem with a Lenovo Yoga 7 2-in-1 14AHP9 (also AES pen with Arch Linux).
I did not find a solution other than restarting the laptop. The behavior seems to get better when enabling full preemptive in the kernel parameters, but it still exists.

Jannis161, 2026-09-29:
Adding a report from a Lenovo ThinkPad L13 Yoga (20R5000KGE) — same symptom as here, and it matches the pattern discussed in #342 and #416.

**Device**: Lenovo ThinkPad L13 Yoga (20R5000KGE)
**Digitizer**: USB 056a:51f9 "Wacom Pen and multitouch sensor" (AES, garaged stylus)
**OS/kernel**: Fedora 44, Linux 7.2.7-200.fc44.x86_64, cmdline already contains `preempt=full`
**Affected stylus**: Wacom Bamboo Ink (external AES pen)
**Clean stylus**: Lenovo ThinkPad Pen Pro (garaged AES pen)

### Summary

With the Bamboo Ink the pen writes fine for a short time, then input develops long lag, then strokes/clicks are swallowed almost entirely; behavior intermittently recovers for a moment. dmesg fills with:

```
wacom 0003:056A:51F9.0004: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
```

200+ occurrences within ~45 minutes of normal use. The log is quiet when no pen is anywhere near the screen (0 occurrences over 90 s hands-off), so the state is tied to actual pen activity, not phantom proximity from the garaged pen.

### The failure follows the pen, not the tablet

A/B alternation (as suggested in #342): when the Bamboo Ink is lagging, switching directly to the garaged ThinkPad Pen Pro works flawlessly every time — no lag, no dropped input — and switching back to the Bamboo Ink brings the failure back immediately. Repeated several times.

So the sensor/driver is not left in a globally broken state by the Bamboo Ink; only that pen's reports trigger the idleprox churn. A fresh alkaline AAAA cell in the Bamboo Ink changes nothing, so it is not plain battery depletion (marginal capacitive coupling is my current suspect).

### What does NOT help / is already ruled out

- **USB re-authorization of 1-8** (`echo 0/1 > authorized`) resets the sensor and the idleprox spam stops — but with the Bamboo Ink the hang returns within ~2 minutes of continued use. Resettable state, same trigger.
- **`preempt=full`** on the kernel cmdline (the workaround mentioned in this issue): already set here, the hang recurs anyway.
- **Protocol mode of the Bamboo Ink** (hold both side buttons ~2 s toggles AES/MPP): the working mode is AES; in MPP mode the panel detects nothing at all. Not a wrong-protocol-mode artifact.
- **Not the event-queue-overflow variant of #342**: no `wacom_wac_queue_insert: kfifo has filled` messages anywhere in the log.

### Willingness to test

Per #416 you offered to provide a driver with the idleprox timer removed: yes, I will gladly build and test it on this machine. This ThinkPad L13 Yoga is available as a test bed (it has one pen that reliably triggers the bug and one that never does — a clean A/B pair for evaluating a candidate fix).

### Sample log

```
Sep 29 18:33:56 kernel: wacom 0003:056A:51F9.0004: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Sep 29 18:33:59 kernel: wacom 0003:056A:51F9.0004: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
Sep 29 18:34:00 kernel: wacom 0003:056A:51F9.0004: wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out.
... (continuous, correlates 1:1 with the lag phases; silent when the pen is not in use)
```