linuxwacom/input-wacom #432: Wacom module not loaded for "key rejected by service" in Secure Boot

Vittucci, 2024-07-08:
I have a Wacom One and I am trying to install the input-wacom drivers (last version, 1.2) in Ubuntu 22.04 on a Dell Precision following the [instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source).

The installation went fine, the module is loaded and the tablet works well if I disable "Secure Boot".

However, when Secure Boot is enabled, the module is not loaded and the tablet does not work.
This is the output of the command `sudo modprobe wacom`:

`modprobe: ERROR: could not insert 'wacom': Key was rejected by service`

This is in spite of the fact that I followed the instructions at the page [Secure Boot](https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot).

`sudo update-secureboot-policy --new-key`
`sudo mokutil --import /var/lib/shim-signed/mok/MOK.der`

`SKIP: /var/lib/shim-signed/mok/MOK.der is already enrolled`

I have also tried to solve the issue by running the following commands:

`if test -x ./autogen.sh --with-signing-key=/root/MOK.priv --with-signing-cert=/root/MOK.der; then ./autogen.sh --with-signing-key=/root/MOK.priv --with-signing-cert=/root/MOK.der; else ./configure --with-signing-key=/root/MOK.priv --with-signing-cert=/root/MOK.der; fi && make && sudo make install || echo "Build Failed"`

and 

`if test -x ./autogen.sh --with-signing-key=/var/lib/shim-signed/mok/MOK.priv --with-signing-cert=/var/lib/shim-signed/mok/MOK.der; then ./autogen.sh --with-signing-key=/var/lib/shim-signed/mok/MOK.priv --with-signing-cert=/var/lib/shim-signed/mok/MOK.der; else ./configure --with-signing-key=/var/lib/shim-signed/mok/MOK.priv --with-signing-cert=/var/lib/shim-signed/mok/MOK.der; fi && make && sudo make install || echo "Build Failed"`

Also in these cases the installation process goes fine, but the issue is still there.

Any suggestion?

Thanks.

meta4icalbot, 2024-08-10:
Same here - Fedora 6.10.3-200.fc40.x86_64
[image]
This missing signing_key.pem file seems to be the issue but my research has not turned up anything useful so far. Following the [Secure Boot](https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot) guide seemed to make no difference.

kenimou, 2025-03-21:
I am having the same issue. Did either of you find a solution?

Joshua-Dickens, 2025-04-09:
> I am having the same issue. Did either of you find a solution?
> 
> For reference I am using Mint 22.1. My wacom is CTL-6100WL.
@kenimou 

Taking a look into this and I was able to replicate the issue on a local test system.

Just wanting to check some details while I work on a solution, if you disable secure boot does your tablet and built module work as expected? And did you follow the secure boot guide as it is currently written?

Joshua-Dickens, 2025-04-10:
Alright it looks like our secure boot instructions are out of date. While we work on getting the page up to date, here's how I got this working on Fedora.

First follow this [link](https://docs.fedoraproject.org/en-US/fedora/rawhide/system-administrators-guide/kernel-module-driver-configuration/Working_with_Kernel_Modules/#sect-generating-a-public-private-x509-key-pair) to generate the proper key pair then enroll it on your system. Then follow our build instructions but remove the sudo make install portion of the command provided in our instructions. Go into the directory where the .ko file was generated and use the following command to sign the file.

]# /usr/src/kernels/$(uname -r)/scripts/sign-file \
 sha256 \
 my_signing_key.priv \
 my_signing_key_pub.der \
 wacom.ko

Replacing the .priv and .der in the command for wherever you've played your private and public keys generated by the above process. Then just go back into the input-wacom directory and run sudo make install and reboot like normal.

This process should be similar for Ubuntu but I haven't been able to test that yet so for now check out this [link](https://wiki.ubuntu.com/UEFI/SecureBoot/Signing) which seems to be basically the same process just with a different command.

Joshua-Dickens, 2025-04-15:
I've updated our secure boot page with processes that I tested and got working. I recommend trying it out and providing feedback if y'all are still having this issue to make sure the updated process works on a variety of systems.

Pinglinux, 2025-05-08:
Anyone of you, @kenimou @meta4icalbot and @Vittucci, tested Josh's fix? If we don't hear from you, I am going to close the issue.

closed 2025-06-21 by Pinglinux