linuxwacom/input-wacom #433: Pen is not working in fullscreen

cankurttekin, 2024-07-18:
I am using Lenovo Yoga 7i 14irl8, 
[image]

when i open apps in fullscreen mode pen only works if i dont lift up, as long as i lift up the pen once it does not take inputs if i dont quit from fullscreen. 
I tried observing events with `libinput debug-events --verbose` and it detecs my pen inputs even on fullscreen mode but none of my application takes it.
 event5   TABLET_TOOL_AXIS        +2.203s		192.35*/90.68*	tilt: 0.00/0.00	pressure: 0.00

libinput list-devices output:
```

`Device:           Wacom HID 5362 Pen
Kernel:           /dev/input/event5
Group:            6
Seat:             seat0, default
Size:             302x188mm
Capabilities:     tablet 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   none
Rotation:         n/a

Device:           Wacom HID 5362 Finger
Kernel:           /dev/input/event6
Group:            6
Seat:             seat0, default
Size:             302x188mm
Capabilities:     touch 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   n/a
Rotation:         0.0
`

```

flynx, 2024-07-23:
Hi,

Got exactly the same issue, with mostly the same stack:
Lenovo ThinkPad X1 Yoga 3rd
Fedora Linux 40 (Workstation Edition)
Linux 6.9.8-200.fc40.x86_64 x86_64
Wayland
GNOME Shell 46.3.1

libinput list-devices:
```
libinput-config: configuring device 'Wacom Pen and multitouch sensor Finger'
Device:           Wacom Pen and multitouch sensor Finger
Kernel:           /dev/input/event5
Group:            5
Seat:             seat0, default
Size:             309x174mm
Capabilities:     touch 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   n/a
Rotation:         0.0

libinput-config: configuring device 'Wacom Pen and multitouch sensor Pen'
Device:           Wacom Pen and multitouch sensor Pen
Kernel:           /dev/input/event7
Group:            5
Seat:             seat0, default
Size:             309x174mm
Capabilities:     tablet 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   none
Rotation:         n/a
```

cankurttekin, 2024-07-23:
i am not sure but this is probably gnome and wayland thing. i have dual boot with gnome 42 and it does work perfectly on that. we might open this issue on gnome vcs

flynx, 2024-08-05:
@OlaszPL everything worked for me several updates back as well, but I can't exactly say when things break and on what exact package.

cankurttekin, 2024-08-05:
> @OlaszPL everything worked for me several updates back as well, but I can't exactly say when things broke and on what exact package update.

Is it possible to you to find what is latest working version?

OlaszPL, 2024-08-06:
I downgraded gnome-shell to 46.0-1 and libwacom to 2.10.0-1 but it didn't work.

cankurttekin, 2024-08-06:
Yeah i also tried that. It works on 44, but i am not sure about 45, will chek that.

flynx, 2024-08-06:
I have everything running on Gnome 45.5 but that's not a clean test -- different hardware.

cankurttekin, 2024-09-13:
> https://gitlab.gnome.org/GNOME/mutter/-/issues/3616 This issue was closed 2 weeks ago, so we just have to wait for a new mutter release. It appears that this fix is included in mutter 47 that would probably be released with Gnome 47.

thanks for the info

closed 2024-09-26 by cankurttekin