linuxwacom/input-wacom #437: Lenovo Precision Pen 2 buttons not mapped

njalooo, 2024-08-26:
Hey, i have a Lenovo Precision Pen 2, which shows up as a "Wacom Pen and multitouch sensor Pen (event7)" in libinput. It seems to connect to the laptop by communicating with the touchscreen, since inputs only register after a certain distance. Internally it shows up as an USB device. 
Under /dev/input/ there is no wacom* symbolic link beeing generated, as indicated by https://wiki.archlinux.org/title/Graphics_tablet Section 5.3.1, hence this issue.
I am running Linux Manjaro, Wayland, KDE Plasma, on a https://wiki.archlinux.org/title/Lenovo_Yoga_7i.
`ls /dev/input/by-id` Returns:` usb-Wacom_Co._Ltd._Pen_and_multitouch_sensor-event-if00  usb-Wacom_Co._Ltd._Pen_and_multitouch_sensor-event-mouse  usb-Wacom_Co._Ltd._Pen_and_multitouch_sensor-mouse`
BTN_TOOL_PEN shows up when hovering, and moves the mouse. Touching the screen returns BTN_TOUCH, and does left-click as intended. Pressing the lower button returns BTN_TOOL_RUBBER, which has no effect, and i can't seem to map it anywhere. Pressing the upper button (returns BTN_STYLUS) is supposed to do a right click, when combined with BTN_TOUCH, but has no effect. Tilt x and tilt y, as well as pressure do register in libinput.
The device doesn't show in KDE Plasmas System Settings.
Is there any way to map this?

RayJW, 2024-10-10:
It seems you have multiple problems. With the next update, which will arrive some time, most of them should be fixed once I can work with the developers to finish the work [here](https://github.com/linuxwacom/libwacom/issues/774). At that point, you will be able to configure the pen and one of its buttons.

Your second issue, however, "can" not be resolved. This is just how styli behave with an eraser button. This button can only be used for erasing and not be remapped. You can read more [here](https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/2279#note_1641232).

Pinglinux, 2025-02-25:
Thank you @RayJW for your support!