linuxwacom/input-wacom #438: KDE panel icons remain sticked to the pointer and windows slip away when moved not from titlebar

aldolat, 2024-08-28:
Hello,
I am experiencing two strange behaviours when using my One by Wacom (M), ID 056a:037b.
I use KDE (current version) on Arch Linux, daily updated.

These are the Wacom packages installed on my system:

```
extra/libwacom 2.12.2-1
extra/wacomtablet 6.1.4-1 (plasma)
extra/xf86-input-wacom 1.2.2-1
```

This is the result of the command `modinfo`:

```
$ modinfo wacom | grep version
version:        v2.00
srcversion:     4B1D5D49AF6632F92E5A456
```

# First bug

When clicking an icon on the KDE panel, this icon remain sticked to the cursor. Think as if I had never released the click, so it's like a click and drag. In this way the panel is totally unusable.

To explain this behaviour better, I made this short video: https://i.imgur.com/ZmaPpZM.mp4

# Second bug

The other problem occurs when trying to move a window. If I click on the window title bar, no problem. If I pick the window in location different from the title bar, the window slips away under the cursor to an arbitrary location. No problem when moving the windows around using the ALT+clic technique.

No problem when using Dolphin to move files around with my tablet.

I made some videos to better explain it.

Using Dolphin: https://i.imgur.com/EE7b9I8.mp4
Using Ark: https://i.imgur.com/FxdQA09.mp4
Using Kreversi: https://i.imgur.com/ga7qWLl.mp4

Thanks in advance.

aldolat, 2024-10-12:
I just updated my Arch to KDE Plasma 6.2 and bug seems resolved. I will test it more and let you know.

Pinglinux, 2025-02-25:
A few months passed. I assume no news is good news.

closed 2025-02-25 by Pinglinux