linuxwacom/input-wacom #440: [Thinkpad Yoga Gen8, 6.11 regression] Pen is always in eraser mode

inducer, 2024-09-18:
As of kernel 6.11 (rc3 is the earliest I know is affected, 6.11 release is, too, 6.9 definitely not), I have the problem that multiple pens I use, including the "Lenovo Precision Pen 2" and the "Dell Active Pen PN579X" seem to always be in eraser mode when used with the Wacom touchscreen device in my machine. libinput reports this as

```
/dev/input/event7:	Wacom HID 5344 Pen
/dev/input/event8:	Wacom HID 5344 Finger
```

[Here](https://gist.github.com/inducer/ce70de26891bd125c7fb1c6402f1c23b) is a libinput event log covering my just bringing the pen into proximity and removing it, my finger is nowhere near the eraser button. Yet still
```
$ grep -i rubber /tmp/evtlog.txt
    #   Event code 321 (BTN_TOOL_RUBBER)
    - [  0,      0,   1, 321,       1] # EV_KEY / BTN_TOOL_RUBBER           1
    - [  0, 687022,   1, 321,       0] # EV_KEY / BTN_TOOL_RUBBER           0
```

libinput is at 1.26 as installed by Debian, but this seems to be a kernel issue; it appears and disappears depending on the kernel version.

RayJW, 2024-10-07:
I reported the issue to libinput as I was unsure where, what was breaking. Seems like this might be the more appropriate place. I'll leave [this](https://gitlab.freedesktop.org/libinput/libinput/-/issues/1041) here in case the additional logs help, and will close my issue if it is confirmed that this is a kernel driver problem.

RayJW, 2024-10-07:
> I suspect the issue might be caused by a recent simplification to our tool type logic.
> 
>     1. Please install the "git" command-line tool.
> 
>     2. Run `git clone https://github.com/jigpu/input-wacom.git -b fix-440` to get a copy of the driver with a potential fix.
> 
>     3. Follow the instructions on our [Installing input-wacom from source](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) wiki page (building the copy of "input-wacom" from step 2 rather than downloading the most recent release).
> 
> 
> After you build and install the driver, reboot and run `grep "" /sys/module/wacom*/version` to verify the driver version. It should be `v2.00-1.2.0.35.gd6fa49e`. Check to see if the pen works normally again.

I just verified that this revert does indeed fix the problem. Everything is working as it did with kernel version 6.11 now :)

GabriF, 2024-10-07:
IdeapadFlex 14ABR8 same problem. Currently using linux-lts kernel 6.6.54 to make pen work correctly.

jigpu, 2024-10-07:
Thanks, @RayJW . I'll work with the other devs to get this fixed upstream.

jigpu, 2024-10-08:
@RayJW Could you please test an updated version of the fix for me? We decided to try a more focused approach rather than simply reverting the whole patch. You can get the new version by running `git clone https://github.com/jigpu/input-wacom.git -b fix-440b` and then rebuilding/reinstalling. This patch should report its version as `v2.00-0.49.0.66.gf169e18`.

If this updated version still works well, I'll send it upstream to the LKML so that it gets fixed ASAP.

RayJW, 2024-10-08:
> @RayJW Could you please test an updated version of the fix for me? We decided to try a more focused approach rather than simply reverting the whole patch. You can get the new version by running `git [...]

Hmm, for some reason when I run `grep "" /sys/module/wacom*/version` it reports `v2.00-1.2.0.36.g7fe0cda` but since that does differ from the previous fix I'll assume that this is correct. I can confirm with the driver from branch `fix-440b` the pen works as expected, and as it did with the 6.10 kernel.

jigpu, 2024-10-09:
Thanks for testing. I got the wrong version number for some reason. The version that you listed looks correct (it has the expected commit hash at the end, unlike the number I posted). I'll go ahead and get the changes sent off to the LKML.

inducer, 2024-10-09:
Just to confirm: I've now also tested the [patch](https://github.com/linuxwacom/input-wacom/compare/master...jigpu:input-wacom:fix-440b.diff) (by applying it directly on top of 6.11.2), and I'm happy to report that this seems to address the issue. Thanks for putting together a fix!

jigpu, 2024-10-11:
Patch has been accepted upstream for Linux 6.12. It should also be automatically backported to stable kernels in the future once its a bit further along the LKML processes.

I've pushed the fix to the master branch of input-wacom, so you should also get the fix when building from source.

closed 2024-10-11 by jigpu

RayJW, 2024-10-11:
> Patch has been accepted upstream for Linux 6.12. It should also be automatically backported to stable kernels in the future once its a bit further along the LKML processes.
> 
> I've pushed the fix to the master branch of input-wacom, so you should also get the fix when building from source.

Thank you for the great work! I assume master doesn't differ from the patch on branch `fix-440b`, so reinstalling isn't really necessary?