linuxwacom/input-wacom #443: Bamboo CTH-470 crashing GNOME on Ubuntu 24.04

JarrettR, 2024-09-29:
I'm using my Bamboo in Blender. I can use it for about 30-40 minutes generally, then the stylus buttons start interacting strangely for a minute (as if I'm holding one of the function buttons) before crashing GNOME and dumping me back to a login screen. This was not happening before the upgrade, in Ubuntu 22.04.

This seems similar to [this issue](https://github.com/linuxwacom/xf86-input-wacom/issues/327), but that issue was fixed and merged into mutter 46.0. Likewise, [this](https://www.reddit.com/r/linuxquestions/comments/1cy700q/wacom_tablet_crashes_gnome/) seems to be similar behaviour.

Versions:
Ubuntu 24.04 LTS
GNOME 46.0
mutter 46.2
/sys/module/wacom/version v2.00

`dmesg` after the crash only displays information after the crash, starting again at timestamp 0 so not particularly helpful. Other than that, I'm not sure what logs or troubleshooting steps would help.

jigpu, 2024-10-17:
It might be helpful to gather some of the debug info about this crash. I just updated our wiki with some instructions on how to gather information about crashes. Please follow the instructions here https://github.com/linuxwacom/input-wacom/wiki/Getting-Crash-Dumps-on-Ubuntu and reply back with a Stacktrace for review.

Also, are you running GNOME in an X11 or Wayland session?

JarrettR, 2024-12-13:
Thank you for the instructions! That's incredibly helpful.

Blender updated on me and I'm no longer able to trigger the crash. I will reopen if that changes.

closed 2024-12-13 by JarrettR