linuxwacom/input-wacom #445: Tilt sensitivity not working correctly on wacom intuos pro M using bluetooth

Ulisesrec, 2024-10-17:
As said in  the title i'm using a Wacom Intuos pro on bluetooth mode and the tilt sensitivity does not work correctly, it does work while wired.
After a little bit of testing looking at the pen inputs while wired and with bluetooth connection using xinput, i've seen that while it is wireless the tilt values can't go over to negatives values (tilted upwards and to the left), is there any solution for that?

Versions:
Ubuntu 22.04 LTS
GNOME 42.9
/sys/module/wacom/version v2.00-input-wacom-1.3.0

pablogila, 2024-10-20:
Same problem here with a Wacom Intuos CTL 4100 (with cable)...

jigpu, 2024-10-24:
I've been able to reproduce this issue and believe I've found a solution. Please 

1. Please install the "git" command-line tool.
2. Run git clone `https://github.com/jigpu/input-wacom.git -b fix-445` to get a copy of the driver with a potential fix.
3. Follow the instructions on our [Installing input-wacom from source](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) wiki page (building the copy of "input-wacom" from step 2 rather than downloading the most recent release).

After you build and install the driver, reboot and check to see if the pen works normally again. If it does not, run `grep "" /sys/module/wacom*/version` and post its value back here so that I can make sure you're running the correct version of the driver.

Ulisesrec, 2024-10-26:
This worked! Thank you very much!
Marking this as completed now.

closed 2024-10-26 by Ulisesrec

sbrl, 2024-11-18:
Just want to chime in here because I had to do some extra steps to get this to work.

To outline the full steps I did, I first ran through the steps outlined above:

```bash
git clone https://github.com/jigpu/input-wacom.git -b fix-445
if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
sudo reboot
```

....but this didn't help, as I had the wrong version of the `wacom` kernel module:

```bash
$ grep "" /sys/module/wacom*/version
v2.00
```

...then I tried reloading the kernel module:

```bash
sudo rmmod wacom
sudo modprobe wacom
```

.....but this didn't not shift the version number either:

```bash
$ grep "" /sys/module/wacom*/version
v2.00
```

What I ended up doing was removing the `xserver-xorg-input-wacom` package, and then running through the above installation steps again.

Then I cycled the `wacom` kernel module:

```bash
sudo rmmod wacom
sudo modprobe wacom
```

....and then this worked!

```bash
$ grep "" /sys/module/wacom*/version
v2.00-1.2.0.37.g2c27caa
```

In hindsight, the first reboot wasn't really necessary when installing - I could have reloaded the kernel module manually.

An issue I did encounter though is that since I'm a leftie I have a custom script for changing the orientation of my tablet (sadly I can't find the right setting to update the touch - only the pen :-/). To fix this, I simply downloaded the `.deb` package for the package I just removed and extracted the `xsetwacom` binary manually:

```bash
sudo apt download xserver-xorg-input-wacom
dpkg -x "xserver-xorg-input-wacom_1%3a1.2.0-1ubuntu2_amd64.deb" . # OR: ar xv "xserver-xorg-input-wacom_1%3a1.2.0-1ubuntu2_amd64.deb"
tree
mv usr/bin/xsetwacom /usr/local/games # picked /usr/local/games because in my $PATH it's AFTER everything else, just in case I forget about it and it causes an issue
```

Related scripts:

- `extract`: <https://gitlab.com/sbrl/bin/-/blob/master/extract>
- `setup-intuos-pro-s.sh` (my setup script 'cause I'm a leftie): <https://gitlab.com/sbrl/bin/-/blob/master/scripts/setup-intuos-pro-s.sh?ref_type=heads>
     - `setup-intuos-pro-s-listen.py` (watcher script that starts on login to auto-apply settings on device connect): <https://gitlab.com/sbrl/bin/-/blob/master/scripts/setup-intuos-pro-s-listen.py?ref_type=heads>

Using the tablet tester in Krita I now correctly see negative numbers as I move the brush around and tilt it 🌟 ❤️

I do note that there are various different symbols next to brushes. It seems like the ones with the 🔁 (rotating arrows) symbol only support the rotational angle of the brush, while the ones with the ✏️ pencil-like symbol also support changing what the brush looks like when you tilt the pen at different angles relative to the graphics tablet.

System details:

[... 273 lines cut ...]
     GL_ARB_fragment_shader_interlock 
     GL_MESA_framebuffer_flip_y 
     GL_ARB_shader_draw_parameters 
     GL_ARB_texture_stencil8 
     GL_ARB_enhanced_layouts 
     GL_ARB_texture_mirror_clamp_to_edge 
     GL_ARB_texture_query_levels 
     GL_ARB_uniform_buffer_object 
     GL_EXT_vertex_attrib_64bit 
     GL_ARB_texture_border_clamp 
     GL_ARB_texture_query_lod 
     GL_AMD_conservative_depth 

QPA OpenGL Detection Info 
  supportsDesktopGL: true 
  supportsOpenGLES: true 
  isQtPreferOpenGLES: false 
  Detected renderers: 
    (Supported) Mesa Intel(R) Graphics (ADL GT2) (4.6 (Core Profile) Mesa 24.0.9-0ubuntu0.2) 
    (Supported) Mesa Intel(R) Graphics (ADL GT2) (OpenGL ES 3.2 Mesa 24.0.9-0ubuntu0.2)  

useBufferInvalidation (config option): false
</code></pre>

</details>

sbrl, 2024-11-21:
Update: `xsetwacom` has broken now, which is a pain. Need to find a solution for that to make my tablet actually useable, but don't have time until tomorrow to investigate.

sbrl, 2024-11-22:
Okay, so the problem was that by removing the `` package, I also removed `xf86-input-wacom`, which is not the same thing as `input-wacom`. One is the kernel module (this one), and the other is the X11 driver, as far as I understand it.

So compiling & installing the xf86-input-wacom repo from source fixed it for me.

I've written my entire process up into a blog post which I need to finish writing - I'll edit this comment tomorrow with a link to it since the full write-up is far too long for a GitHub comment and I've already hogged this issue enough.

I summarise the process below:

-----------

Instructions for this are partially sourced from <https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver#building-with-autotools>.

First, install dependencies:

```bash
#shellcheck disable=SC2046
sudo apt-get install autoconf pkg-config make xutils-dev libtool xserver-xorg-dev$(dpkg -S $(which Xorg) | grep -Eo -- "-hwe-[^:]*") libx11-dev libxi-dev libxrandr-dev libxinerama-dev libudev-dev
```

Then, clone the git repository and checkout the latest release:

```bash
git clone https://github.com/linuxwacom/xf86-input-wacom.git
cd "xf86-input-wacom";
git tag; # Pick the latest one from this list
git switch "$(git tag | tail -n1)"; # Basically git switch TAG_NAME
```

It should be at the bottom, or at least that's what I found. For me, that was `xf86-input-wacom-1.2.3`.

Then, to build and install the software from source, run these 2 commands one at a time:

```bash
set -- --prefix="/usr" --libdir="$(readlink -e $(ls -d /usr/lib*/xorg/modules/input/../../../ | head -n1))"
if test -x ./autogen.sh; then ./autogen.sh "$@"; else ./configure "$@"; fi && make && sudo make install || echo "Build Failed"
```

Now you should have the X11 side of things installed. In my case that includes `xsetwacom`, the (questionably designed) CLI for managing the properties of connected graphics tablets. See my previous comment if that isn't the case for you.

Finally, you need to make X11 like the kernel driver. It's just a single config file. Install it like this:

```bash
mkdir -p /etc/X11/xorg.conf.d/;
sudo curl -sSv https://raw.githubusercontent.com/linuxwacom/xf86-input-wac...efs/heads/master/conf/70-wacom.conf -o /etc/X11/xorg.conf.d/70-wacom.conf
```

Just case things are moved around as I've seen happen in far too many comments and tutorials with broken links before, the direct link to the exact commit of this file I used is: <https://github.com/linuxwacom/xf86-input-wacom/blob/47552e13e714ab6b8c2dcbce0d7e0bca6d8a8bf0/conf/70-wacom.conf>

Then finally reboot and everything should Just Work™!

jigpu, 2024-12-18:
@sbrl and anyone else who comes across this thread in the future, I strongly advise against those particular installation steps if you are "just" trying to get the input-wacom driver working. The "input-wacom" and "xf86-input-wacom" drives are completely separate things; you will never need to replace the latter to get the former working. (In extremely rare instances you /do/ have to install updated versions of both packages, but if you're /only/ trying to update input-wacom it is not required!)

I suspect that what was happening is that the `modprobe` program will not necessarily make use of the newly-installed driver without a reboot. If it still is not being used after a reboot, then there is a problem with our build scripts that we'll need to address (specifically: something weird will be going on with depmod...).

Ulisesrec, 2024-12-18:
Oh, eh wait i haven't used the tilt in sometime so i didn't notice, but i just checked and It is Broken again, do i have to update It? (did It broke bc It maybe updated itself? It is now v2.00) Or i have to install the fix again?

jigpu, 2024-12-18:
Your system more than likely installed a kernel update that broke things. The driver built by input-wacom is tied to a specific kernel version, so it will "stop working" when an update comes along. If you rebuild and reinstall input-wacom, things should start working again. (Unfortunately you'll have to rebuild/reinstall for every kernel update until your distro ships a version of our driver that includes the fix).

Ulisesrec, 2024-12-18:
Oooh i see, thanks!
I do have to install this specific one right? The fix-445
And well it may be kinda of annoying having It break but it's not too much of a hassle

jigpu, 2024-12-18:
The fix is not in a released version of input-wacom yet, but it _has_ been merged onto its master branch. You don't need to rely on my own personal "fix-445" branch anymore (though that does still also work)

Instead of using `git clone https://github.com/jigpu/input-wacom.git -b fix-445` I recommend using `git clone https://github.com/linuxwacom/input-wacom.git` to get the most up-to-date version of input-wacom.

The fix should trickle its way into distributions soon, at which point you should be able to stop relying on input-wacom.

Ulisesrec, 2024-12-20:
Okay, thanks!