linuxwacom/input-wacom #448: Wacom tabled needs to be connected to MAC or Windows before it works on Ubuntu

leli-ilp, 2024-11-11:
Hi unsung heroes of the Linux Wacom community,

We noticed that in order for a Wacom tablet to work on an Ubuntu distribution we needed to first plug it into a Windows or a MAC client before it's properly working with all fancy features.

**Test client**
Distribution: Ubuntu 22.04
Kernel: 5.15.0-125-generic
Root drive: /dev/nvme0n1
Module version
```bash
# modinfo wacom
filename:       /lib/modules/5.15.0-125-generic/extra/wacom.ko
license:        GPL
description:    USB Wacom tablet driver
author:         Vojtech Pavlik <vojtech@ucw.cz>
version:        v2.00-1.3.0
```

**Scenario**

Connecting a wacom tablet to a Ubuntu client results in the following logs
```bash
Nov 11 15:27:14 ubuntu-client kernel: scsi 2:0:0:0: Direct-Access     Wacom    Intuos Pro L     1.53 PQ: 0 ANSI: 2
Nov 11 15:27:14 ubuntu-client kernel: sd 2:0:0:0: Attached scsi generic sg0 type 0
Nov 11 15:27:14 ubuntu-client kernel: sd 2:0:0:0: [sda] 65536 512-byte logical blocks: (33.6 MB/32.0 MiB)
Nov 11 15:27:14 ubuntu-client kernel: sd 2:0:0:0: [sda] Write Protect is on
Nov 11 15:27:14 ubuntu-client kernel: sd 2:0:0:0: [sda] Mode Sense: 00 00 80 00
Nov 11 15:27:14 ubuntu-client kernel: sd 2:0:0:0: [sda] Asking for cache data failed
Nov 11 15:27:14 ubuntu-client kernel: sd 2:0:0:0: [sda] Assuming drive cache: write through
Nov 11 15:27:14 ubuntu-client kernel: sd 2:0:0:0: [sda] Attached SCSI removable disk
Nov 11 15:27:14 ubuntu-client upowerd[1830]: treated changed event as add on /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:056A:0358.0010/power_supply/wacom_battery_5
Nov 11 15:27:15 ubuntu-client systemd-udevd[25424]: sda: Process '/usr/bin/unshare -m /usr/bin/snap auto-import --mount=/dev/sda' failed with exit code 1.
Nov 11 15:27:15 ubuntu-client kernel: /dev/sda: Can't open blockdev
Nov 11 15:27:15 ubuntu-client udisksd[1069]: Mounted /dev/sda at /media/mrbean/Wacom Start on behalf of uid 1656
Nov 11 15:27:15 ubuntu-client dbus-daemon[2598]: [session uid=1656 pid=2598] Activating service name='org.gnome.Shell.HotplugSniffer' requested by ':1.31' (uid=1656 pid=2814 comm="/usr/bin/gnome-shell " label="unconfined")
Nov 11 15:27:15 ubuntu-client dbus-daemon[2598]: [session uid=1656 pid=2598] Successfully activated service 'org.gnome.Shell.HotplugSniffer'
```
Noticed that the device appears under fdisk -l

```bash
Disk /dev/sda: 32 MiB, 33554432 bytes, 65536 sectors
Disk model: Intuos Pro L    
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot      Start        End    Sectors  Size Id Type
/dev/sda1       4294967295 8589934589 4294967295    2T ff BBT
/dev/sda2       4294967295 8589934589 4294967295    2T ff BBT
/dev/sda3       4294967295 8589934589 4294967295    2T ff BBT
/dev/sda4       4294967295 6854241533 2559274239  1.2T ff BBT
```

Device is detected but only the basic functionality is there, pressure sensitivity isn't working for example.

We now connect it to a MAC with Wacom Center installed.
Wait for it to be recognized, disconnect it from the MAC and connect it back to the Ubuntu Linux workstation and BOOM, it's now working. 

Journalctl isn't showing the `kernel: /dev/sda: Can't open blockdev` error anymore and fdisk isn't listing it as a device under `/dev/sda`.

Problem now is that we need to plug every wacom board to either a Windows PC or a MAC before handing it out to our Linux users. I was hoping that there's an explanation to why this is happening. 

Best regards

Pinglinux, 2024-12-25:
Can you give us the device's ID and model number?

Pinglinux, 2025-01-03:
A new wiki [page](https://github.com/linuxwacom/input-wacom/wiki/Disable-Intuos-Pro-M-and-L-Storage-Interface) has been created for the above information.

closed 2025-01-03 by Pinglinux