linuxwacom/input-wacom #457: Intuos Pro S (PTK470) dial and buttons do not work

goekce, 2025-04-15:
Kernel version: 6.14.2 (ArchLinux)
I use sway 1.10.1 based on Wayland

I tried both Bluetooth and USB. Pen works, however the buttons and the dial does not have any effect in `wev` (A tool for debugging wayland events on a Wayland window, analagous to the X11 tool xev).

Kernel messages:
```
[ 1444.655448] usb 3-3: new full-speed USB device number 24 using xhci_hcd
[ 1444.782168] usb 3-3: New USB device found, idVendor=056a, idProduct=03f5, bcdDevice= 1.03
[ 1444.782182] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1444.782186] usb 3-3: Product: Wacom Intuos Pro S
[ 1444.782190] usb 3-3: Manufacturer: Wacom Co.,Ltd.
[ 1444.782193] usb 3-3: SerialNumber: 5CH21J1001088
[ 1444.790716] wacom 0003:056A:03F5.000D: hidraw8: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom Intuos Pro S] on usb-0000:00:14.0-3/input0
[ 1444.790766] input: Wacom Intuos Pro S Pen as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:056A:03F5.000D/input/input47
[ 1444.790983] input: Wacom Intuos Pro S Pad as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:056A:03F5.000D/input/input49
[ 1444.792202] wacom 0003:056A:03F5.000E: Unknown device_type for 'Wacom Co.,Ltd. Wacom Intuos Pro S'. Ignoring.
```

Could this be related to the unknown device `000E`?

Pinglinux, 2025-05-08:
Can you use[ hid-tools](https://gitlab.freedesktop.org/libevdev/hid-tools/-/blob/master/README.md?ref_type=heads) to see if you get any output about the tablet buttons and dial from the kernel? If kernel doesn't report any button events, Wayland can't do anything about them.

goekce, 2025-05-08:
`hid-tools` indeed shows activity:

```
# ReportID: 17 / Wacom ExpressKey00: 0 | 0xff0d0911: 0 | 0xff0d0912: 0 | 0xff0d0913: 0 | # | # 
#              | Wacom Button Center: 0 | # 
#              | Wacom TouchRing:   -1 | # | # 
E: 000030.463788 9 11 00 00 00 7f 00 00 00 00
```

 Does this mean that wayland does not listen on these events?

goekce, 2025-05-09:
If `hid-tools` receives the events but `sway` does not receive them, probably because they have to be routed to some actions. I read that some desktop environments like GNOME can configure these events and in case of other environments like `sway`, tools like https://github.com/sezanzeb/input-remapper or https://github.com/OpenTabletDriver/OpenTabletDriver can be used. I believe these tools will help me, otherwise I'll open the issue again. Thanks ☺️

closed 2025-05-09 by goekce