linuxwacom/input-wacom #459: Trying to add touch settings in Gnome

jimafisk, 2025-06-04:
**My Computer Setup:**
- Ubuntu 24.04.2 LTS
- Gnome Shell 46.0
- Wayland

**My Tablet / Pen Display:**
- XP-Pen Artist Pro 16TP
- Driver version 4.0.7 (https://www.xp-pen.com/download/artist-pro-16tp.html)

**Intro:** I'm trying to configure multiple screens in Ubuntu via Settings > Displays and setting Multiple Displays = Join. When doing this, the Wacom Tablet settings (Settings > Wacom Tablet) have been extremely helpful to assign my pen/stylus and eraser to the correct screen.

**Problem:** Unfortunately some of my touch actions are still taking place on the primary screen (my monitor), instead of the tablet. It seems to be treating both screens as one, so if in Settings > Displays I position the screens side-by-side horizontally, touch actions on the left side of my tablet take place on the primary screen (incorrect) but touch gestures on the right side take place on the tablet (correct).

**Solution attempt:** I thought maybe I could extend the Wacom Tablet settings for my device to allow assigning touch to a particular screen, similar to how the pen & eraser are being assigned.

When I run `lsusb | grep XP-Pen` I see:
```
Bus 003 Device 017: ID 28bd:092e XP-Pen 15.6 inch PenDisplay
```

I did a `sudo apt install libwacom-bin` and when I run `libwacom-list-local-devices` it outputs:
```
/dev/input/event30 is a tablet but not supported by libwacom
/dev/input/event28 is a tablet but not supported by libwacom
Failed to find any devices known to libwacom.
```

Since my particular device isn't recognized, I've tried to add a `/usr/share/libwacom/xp-pen-artist-pro-16tp.tablet` file with contents like:

```ini
# XP-Pen
# Artist Pro 16TP
# USB ID: 28bd:092e
# Based on lsusb and /dev/input/event30, /dev/input/event28

[Device]
Name=XP-Pen Artist Pro 16TP
ModelName=Artist Pro 16TP
DeviceMatch=usb:28bd:092e:XP-Pen Artist Pro 16TP;usb:28bd:092e:XP-Pen Artist Pro 16TP Touch
Class=Bamboo
Width=13.4
Height=7.6
IntegratedIn=Display
Styli=@generic-eraser
Layout=xp-pen-artist-pro-16tp.svg

[Features]
Stylus=true
Reversible=true
Touch=true
TouchSwitch=true
Ring=false
Ring2=false
Buttons=0
```

I set the permissions like the other device files `sudo chmod 644 /usr/share/libwacom/xp-pen-artist-pro-16tp.tablet`.

Then I try to update the database `sudo libwacom-update-db` which outputs:

```
New hwdb file: /etc/udev/hwdb.d/66-libwacom.hwdb
Finished, please unplug and replug your device
```

Finally I unplug and replug the device (also tried rebooting), but it doesn't seem to add any additional touch options in Settings > Wacom Tablet.

**Questions:** 
1. Am I following the right process to add support for a new device?
2. Is my `xp-pen-artist-pro-16tp.tablet` file configured incorrectly?
3. Does Gnome even support setting touch in the GUI?

Thank you! I appreciate any guidance in the right direction :).

Penguin-Guru, 2025-06-12:
I do not know the answers to your questions but I will share a couple thoughts in case they help.

1. There are a few ways of implementing multi-monitor support in X11. It is possible that Gnome's approach spans all "output devices" (i.e. video monitors) across each X11 "screen". Some window managers, like i3, assign each "output device" to a separate "screen". I don't know if this could be related to your issue.
2. `man 4 wacom` mentions options (e.g. "TopX") that seem like they could allow you to specify boundaries for your touch input area. I have no idea whether this would have any effect on the visual effects (which I assume originate from some middleware farther up the event chain).

If any of that is wrong, hopefully someone will correct me. :)

Pinglinux, 2025-07-12:
This project is for downstream kernel patches. Please discuss your request at libwacom project.

closed 2025-07-12 by Pinglinux