linuxwacom/input-wacom #465: There's no display on the Wacom One 13.3 on Bazzite

PlatinumLucario, 2025-08-16:
My friend was using the Wacom One 13.3 on their PC, which runs on the latest Bazzite GNOME image (based on Fedora Silverblue), they were able to get the pen and the touch input to work. But it has no input display whatsoever on the tablet.

Their PC doesn't have Thunderbolt 4 support, due to it being a AMD AM5 motherboard, and only has a USB 3.2 Gen 2 port at the back.

We've currently tried using a USB to HDMI cable, but it doesn't work.

So we've been wondering, how do we get the display to show on the Wacom One 13.3 tablet? Is there something that we need to do? Does it need a 3 in 1 cable? Is there a setting we need to change in GNOME Settings?

The only options that comes up when selecting displays in GNOME Settings, is the name of the monitor we use and the options as to whenever to use the same display on all or separately.

Pinglinux, 2025-09-13:
There should be a few ways to connect your tablet to your system's display port. We can't troubleshoot that for you without the exact environment as yours.

PlatinumLucario, 2025-09-27:
> Your display probably doesn't have enough power? A 3-in-1 cable may work. Do you have a USB-c power adapter, the one for laptop not for phone?

The 3-in-1 cable completely fixed the issue for my friend's PC. It was due to the USB-C to HDMI adapter not receiving enough power delivery from the USB-C port. So the 3-in-1 cable was definitely needed. So we're all good now.

closed 2025-09-27 by PlatinumLucario